lathoub/Arduino-AppleMIDI-Library #157: I need 16 simultaneous sessions, but Teensy 4.1 is restarting when I use more than 4 sessions.

AtalibaJunior, 2023-01-04:


AtalibaJunior, 2023-01-04:
Hello, forgive me. I'm new here on github and I'm learning.
I'll paste the code below:

#include <NativeEthernet.h>
#include <AppleMIDI.h>
#include <Chrono.h>

#define delayLed 3
Chrono tempLed;

bool ledActive = 0; 
const int ledPin = 13;

byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
IPAddress ip(192, 168, 0, 20);

APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP1, "MidiNet-01", 5500); 
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP2, "MidiNet-02", 5502);
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP3, "MidiNet-03", 5504); 
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP4, "MidiNet-04", 5506); 
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP5, "MidiNet-05", 5508); 
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP6, "MidiNet-06", 5510);
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP7, "MidiNet-07", 5512); 
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP8, "MidiNet-08", 5514);
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP9, "MidiNet-09", 5516);
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP10, "MidiNet-10", 5518); 
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP11, "MidiNet-11", 5520); 
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP12, "MidiNet-12", 5522);
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP13, "MidiNet-13", 5524); 
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP14, "MidiNet-14", 5526);
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP15, "MidiNet-15", 5528); 
APPLEMIDI_CREATE_INSTANCE(EthernetUDP, RTP16, "MidiNet-16", 5530);

midi::MidiInterface<appleMidi::AppleMIDISession<EthernetUDP>, appleMidi::AppleMIDISettings>* estNET[16] = {&RTP1, &RTP2, &RTP3, &RTP4, &RTP5, &RTP6, &RTP7, &RTP8, &RTP9, &RTP10, &RTP11, &RTP12, &RTP13, &RTP14, &RTP15, &RTP16};

void setup() {
Ethernet.begin(mac, ip);
pinMode(ledPin, OUTPUT);
digitalWrite(ledPin, LOW);
for(int e = 0; e < 16; e++)
{estNET[e]->begin(MIDI_CHANNEL_OMNI);}
}

void loop() {
for(int i=0; i < 16; i++){
   if(estNET[i]->read()){
   digitalWrite(ledPin, HIGH);
   tempLed.restart();
   ledActive = 1;}}
   
   if(ledActive && tempLed.hasPassed(delayLed)){
    digitalWrite(ledPin, LOW);
    ledActive = 0;
   }}

Thanks for listening.

lathoub, 2023-01-05:
For each instance you create, 2 UDP sockets are constructed - so you are looking at **32** UDP sockets to be open. So probably you are exceeding the max amount of sockets you can have open for the ETH PHY of the Teensy 4.1. If you are using the [NativeEthernet library](https://github.com/vjmuzik/NativeEthernet) there is a maximum amount : https://github.com/vjmuzik/NativeEthernet/blob/cdf6b3a2b5d62559a7c76be5b9d3fa60cf9d686b/src/NativeEthernet.h#L36-L40

I'm not questioning your architecture, but having 16 simultaneous AppleMIDI sessions open at one time sounds excessive (especially on a microcontroller).

(you might want to read up on [Github Markdown](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code) on how to put code in a MD message)

AtalibaJunior, 2023-01-05:
Got it. It's explained.
I will make the necessary changes.
Thank you very much for your attention.
Bye.

lathoub, 2023-01-05:
Welcome 🫡
(you created the issue, you close it)

lathoub, 2023-01-06:
@AtalibaJunior : **you** need to _close_ the issue (button below)