lathoub/Arduino-AppleMIDI-Library #162: Unable to install on M1 apple silicon computer

samspencer5991, 2023-10-03:
Upon trying to install this library in PlatformIO, the library won't install, coming with this error: Library Manager: Installing lathoub/Arduino-AppleMIDI-Library @ ^3.2.0
UnknownPackageError: Could not find the package with 'lathoub/Arduino-AppleMIDI-Library @ ^3.2.0' requirements for your system 'darwin_arm64'

It seems like it could be same issue as: https://github.com/earlephilhower/arduino-pico/issues/661

lathoub, 2023-10-04:
Not sure if this issue related to PlatformIO (on Apple M1) or to your related issue on the Pico.

It might be that the this current version of this library (3.3.0) depends on an upcoming new release of the [underlying MIDI library](https://github.com/FortySevenEffects/arduino_midi_library).

lathoub, 2024-03-16:
any comments back @samspencer5991 ?

samspencer5991, 2024-03-19:
After running into this issue I paused plans for integrating the library into a current project. I'll double check when I get a chance to see if there are outstanding build issues.

closed 2026-08-17 by lathoub