lathoub/Arduino-AppleMIDI-Library #163: MIDI Note Off messages dropped

serifpersia, 2023-10-22:
- What board are you using ?
    - `ESP32S2`
- What version of the Arduino IDE are you using ?
    - `example: 1.8.19`
- What Operating System are you using for the DAW
    - [ ]  Windows
    - [ ] rtpMIDI to send MIDI messages from MIDI Device on PC
- Is your problem related to:
    - [ ]  Dropped messages
- How comfortable are you with code ?
    - [ ]  I've done basic projects

## Describe your project and what you expect to happen:

## Describe your problem (what does not work):

## Steps to reproduce

[image]

lathoub, 2023-10-22:
I don't know how to interpret the messages in the log window, as it still contains mapping information.

Are you sing the version via the Arduino library manager, or a direct clone from github? The Arduino library manager is not the latest as this lib is awaiting an update from the base Arduino MIDI lib.

Your basic setup looks OK, but i'm unsure about the Live routing setting.

Regarding latency, have a look at [Latency](https://github.com/lathoub/Arduino-AppleMIDI-Library/wiki/Keeping-Latency-under-control)

serifpersia, 2023-10-22:
I'm not using DAW so picked MIDI Device in that Live Routing section, and it seems it sends the midi data like that, I verified that the issue is not that it happens even if I use DAW(the virtual port rtpMIDI creates). I am using version from library manager in arduino ide, I will try latest from github release. Can latency cause midi messages for some reason only note off to drop? I already have direct midi support on my project but as second option and for testing I'm using midi over network since my midi device is not MIDI USB so I can't use it at the moment with USB Host capable ESP32S2/S3

lathoub, 2023-10-22:
> why would latency cause overwrite of note off messages that are still being processed by appleMIDI?

Latency is (probably) not the cause.

> Playing to many notes too fast causes few note off messages to drop off? 

Yes - but not only Note Off messages (any kind of message could be lost, so not just Note Off).
I have described it [here](https://github.com/lathoub/Arduino-AppleMIDI-Library/wiki/Enlarge-Ethernet-buffer-size-to-avoid-dropping-UDP-packages). Albeit that relates to the use the W5500 chip.
The issue of dropped messages was addressed in the latest release.

Can you provide a buffer of the MIDI messages being send?

You can modify the [test](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/master/test) project to simulate a buffer that contains all the messages that are transmitted at once.

serifpersia, 2023-10-22:
I'm not very familiar with modifying source code files .h .cpp and so on. Is there something I can add to my example code to show you whats going wrong? I only have dropped note off messages.

serifpersia, 2023-10-22:
This is suprising, I created bare bones example and it seems no midi note off message got dropped. 
Buffer empty not played notes last 5 seconds, |
First cycle played 5 notes slowly once, output 5 note on and off's
Second cycle spammed fast 5 notes roughly 5 times nothing got dropped it seems which I find crazy now lol

    21:44:35.987 -> Note On Buffer:
    21:44:35.987 -> Note Off Buffer:
    21:44:35.987 -> Total Note Ons: 0
    21:44:35.987 -> Total Note Offs: 0
    21:44:40.968 -> Note On Buffer:
    21:44:40.968 -> Note On: 28
    21:44:40.968 -> Note On: 21
    21:44:40.968 -> Note On: 23
    21:44:40.968 -> Note On: 24
    21:44:40.968 -> Note On: 26
    21:44:40.968 -> Note Off Buffer:
    21:44:41.003 -> Note Off: 23
    21:44:41.003 -> Note Off: 21
    21:44:41.003 -> Note Off: 26
    21:44:41.003 -> Note Off: 28
    21:44:41.003 -> Note Off: 24
    21:44:41.003 -> Total Note Ons: 5
    21:44:41.003 -> Total Note Offs: 5
    21:44:45.970 -> Note On Buffer:
    21:44:45.970 -> Note On: 28
    21:44:45.970 -> Note On: 21
    21:44:45.970 -> Note On: 26
    21:44:45.970 -> Note On: 24
    21:44:45.970 -> Note On: 23
    21:44:45.970 -> Note On: 28
    21:44:45.970 -> Note On: 24
    21:44:46.003 -> Note On: 26
    21:44:46.003 -> Note On: 23
    21:44:46.003 -> Note On: 21
    21:44:46.003 -> Note On: 28
    21:44:46.003 -> Note On: 26
    21:44:46.003 -> Note On: 21
    21:44:46.003 -> Note On: 24
    21:44:46.003 -> Note On: 23
    21:44:46.003 -> Note On: 28
    21:44:46.003 -> Note On: 26
    21:44:46.003 -> Note On: 24
    21:44:46.003 -> Note On: 23
    21:44:46.003 -> Note On: 28
    21:44:46.003 -> Note On: 21
    21:44:46.003 -> Note On: 26
    21:44:46.003 -> Note On: 24
    21:44:46.003 -> Note On: 23
    21:44:46.003 -> Note Off Buffer:
    21:44:46.003 -> Note Off: 26
    21:44:46.003 -> Note Off: 28
    21:44:46.003 -> Note Off: 24
    21:44:46.003 -> Note Off: 23
    21:44:46.003 -> Note Off: 21
    21:44:46.003 -> Note Off: 28
    21:44:46.003 -> Note Off: 26
    21:44:46.003 -> Note Off: 24
    21:44:46.003 -> Note Off: 21
    21:44:46.003 -> Note Off: 23
    21:44:46.037 -> Note Off: 28
    21:44:46.037 -> Note Off: 26
    21:44:46.037 -> Note Off: 24
    21:44:46.037 -> Note Off: 23
    21:44:46.037 -> Note Off: 28
    21:44:46.037 -> Note Off: 24
    21:44:46.037 -> Note Off: 23
    21:44:46.037 -> Note Off: 26
    21:44:46.037 -> Note Off: 21
    21:44:46.037 -> Note Off: 28
    21:44:46.037 -> Note Off: 26
    21:44:46.037 -> Note Off: 24
    21:44:46.037 -> Note Off: 23
    21:44:46.037 -> Note Off: 21
    21:44:46.037 -> Total Note Ons: 24
    21:44:46.037 -> Total Note Offs: 24

Here is new code

#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiUdp.h>
#include <AppleMIDI.h>

char ssid[] = "Wifi 2.4Ghz"; // your network SSID (name)
char pass[] = "cigani123";    // your network password (use for WPA, or use as key for WEP)

int8_t isConnected = 0;

APPLEMIDI_CREATE_DEFAULTSESSION_INSTANCE();

// Buffers for storing MIDI messages
const int bufferSize = 100;
byte noteOnBuffer[bufferSize];
byte noteOffBuffer[bufferSize];
int noteOnCount = 0;
int noteOffCount = 0;

void setup() {
    Serial.begin(115200); // Initialize the serial monitor
    Serial.println("Booting");

    WiFi.begin(ssid, pass);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.println("Establishing connection to WiFi...");
    }

    Serial.println("Connected to network");

    Serial.print(WiFi.localIP());
    Serial.print(" Port ");
    Serial.print(AppleMIDI.getPort());
    Serial.print(" (Name ");
    Serial.print(AppleMIDI.getName());
    Serial.println(")");

    MIDI.begin();

    AppleMIDI.setHandleConnected([](const APPLEMIDI_NAMESPACE::ssrc_t & ssrc, const char* name) {
        isConnected++;
        Serial.print("Connected to session ");
        Serial.print(ssrc);
        Serial.print(" Name ");
        Serial.println(name);
    });

    AppleMIDI.setHandleDisconnected([](const APPLEMIDI_NAMESPACE::ssrc_t & ssrc) {
        isConnected--;
        Serial.print("Disconnected ");
        Serial.println(ssrc);
    });

    MIDI.setHandleNoteOn([](byte channel, byte note, byte velocity) {
        // Store Note On messages in the buffer
        noteOnBuffer[noteOnCount] = note;
        noteOnCount = (noteOnCount + 1) % bufferSize;
    });

    MIDI.setHandleNoteOff([](byte channel, byte note, byte velocity) {
        // Store Note Off messages in the buffer
        noteOffBuffer[noteOffCount] = note;
        noteOffCount = (noteOffCount + 1) % bufferSize;
    });
}

void printAndClearBuffers() {
    Serial.println("Note On Buffer:");
    for (int i = 0; i < noteOnCount; i++) {
        Serial.print("Note On: ");
        Serial.println(noteOnBuffer[i]);
    }

    Serial.println("Note Off Buffer:");
    for (int i = 0; i < noteOffCount; i++) {
        Serial.print("Note Off: ");
        Serial.println(noteOffBuffer[i]);
    }

    // Summary
    Serial.print("Total Note Ons: ");
    Serial.println(noteOnCount);
    Serial.print("Total Note Offs: ");
    Serial.println(noteOffCount);

    // Clear the buffers and reset the counters
    noteOnCount = 0;
    noteOffCount = 0;
}

void loop() {
    MIDI.read();

    static unsigned long lastPrintTime = 0;
    unsigned long currentTime = millis();

    // Print and clear buffers every 5 seconds
    if (currentTime - lastPrintTime >= 5000) {
        printAndClearBuffers();
        lastPrintTime = currentTime;
    }
}

serifpersia, 2023-10-24:
Just wanted to report using RTOS I created a midiTask to handle midi and I have no more issues. 1ms tick and 2048 memory size seems to do the trick.

    void midiTask(void *pvParameters) {
       while (1) {
    MIDI.read(); // Handle MIDI messages
    vTaskDelay(pdMS_TO_TICKS(1)); // Adjust the delay as needed
      }
   }

closed 2023-10-24 by serifpersia

lathoub, 2023-10-25:
Thanks for the clarification @serifpersia (and well spotted)